Santroller/Santroller #273: [Issue]: App crashing on MacOS

ramossjs, 2025-02-08:
### Please describe your issue.

The program opens normally, I connect my controller and it gets detected, but when I click configure the program crashes, my microcontroller is a Sparkfun Pro Micro 5V, the guitar is already working as a XInput in Windows but im trying to reprogramming to use it on MacOS

### Version

10.2.17

### What operating system are you using?

Mac OS

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2025-02-08:
If you used santroller before it should just work on macos, you don't need to change the programming.
I'm not sure why it's crashing, I don't have much experience with macos to figure that out

ramossjs, 2025-02-09:
i used the old guitar configurator, but xinput isn't supported on macos, that's why i'm trying to reprogramming the guitar

sanjay900, 2025-02-09:
Try following the rescue steps to wipe the config on it, and then programming it again from scratch, it might just be having issues with the old config

sanjay900, 2025-07-05:
Its worth trying the older `10.1.188` version, it seems like the newer builds don't play too nice on some of the arduinos.

closed 2025-07-05 by sanjay900