Santroller/Santroller #274: Need help settings up MIDI adapter

vladborovtsov, 2025-02-10:
### Please describe your request.

Hello, I'm struggling to setup a MIDI-USB adapter. Absolutely no luck.  

What I have by hand: Yamaha DD-75 e-drums kit, a few Pi Picos, U2MIDI Pro Adapter by CME. 

- Santroller Configurator recognizes the U2MIDI Pro Device, but I'm unable to map drums to the appropriate controls. Find note feature just doesnt work. Although, the cable itself is functional. It works just fine with DAW software on PC/MAC. I'm also able to see that drum kit is really sending Note ON/OFF events using MIDI Monitor. 
- Drums are just drums. What is the recommended way to map other controls, like digital pad, start/select, etc in order to be able to sign in into the game and navigate menus? 

### What operating system are you using?

Mac OS

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2025-02-10:
have you tried turning the debounce way up to like 20 or 40?
Depending on the kit find note may just not work, if it triggers too quickly it may not see the event happen

vladborovtsov, 2025-02-11:
Didn't try debounce. Will try it!

As for hub - do you mean I can plug a USB hub into Pico, plug USB MIDI adapter and some usb controller into hub and Santroller will recognize both devices, properly mapping MIDI notes to drums and controller input to controls like ABXY, dPad, etc?

sanjay900, 2025-02-11:
Most of the time, yeah
Some USB hubs don't quite like the emulated USB port but most of them do

vladborovtsov, 2025-02-11:
Omg! I thought it wasn't possible! When I first saw the "X connected devices" this made me think Santroller supports multiple. So I've tried, but it always recognizes only one of the connected devices - either USB MIDI or PS3 gamepad. 

I have two no-name hubs at the moment, but they don't work as expected. 
Maybe you can recommend particular vendor and/or model?
Also should it be USB2/3 hub or USB 2.0 only?

sanjay900, 2025-02-11:
Ive seen people recommend sabrent hubs, usb2 or usb3 hubs work ive been able to use both successfully myself.

sanjay900, 2025-02-11:
Turn down the tempo on your drum brain to the lowest that it goes, that will slow down the clock pulses which are going the be what's causing the problem yeah.

And yeah, if the default midi notes don't work you just disable them and add new ones.

vladborovtsov, 2025-02-11:
I’ve figured out most of the issues. Unfortunately, debouncing between 35 and 40 renders yields the best results, but this performance level is only acceptable for games with easy or medium difficulty.

I also tried using eDrumKit with a MIDI cable connected to the MIDI IN port on the Wii GHWT kit. It works flawlessly, but it’s a setup for a Wii console, which takes up too much space.

I briefly looked into the code at https://github.com/Santroller/Santroller/blob/master/src/pico/midi_host.c. I suspect that what I’m observing is the best performance that Pico can achieve with emulated USB Host. Additionally, the Santroller firmware performs various other tasks behind the scenes. 

I’m considering the possibility of offloading USB MIDI IN to a separate MCU, possibly even another Pico. However, using the built-in USB Host on the Pico instead of PIO might yield better results. I’ve seen the peripheral mode for a secondary Pico, but it’s only intended for direct connections via wires, right?

So, if I create a firmware for the secondary Pico that decodes MIDI and passes it to, for example, I2C, is there any way we could collaborate or integrate it? On the other hand, decoding MIDI into 10 pins and then wiring them directly to the first Pico seems a bit impractical.

sanjay900, 2025-02-11:
Someone else had a dtx400k with this exact problem, and setting the tempo to the lowest it can go on the drum was enough to make it work fine over the emulated USB host. Doing that limits how many clock commands are sent and stops it overwhelming the pico.

At some point I just need to add proper support for normal midi but it's just not something I've had the time to do.

sanjay900, 2025-02-11:
Very odd. Are you playing rb or gh?

vladborovtsov, 2025-02-11:
RB3. I think it must be definitely something to do with MIDI stuff. 

I have another drumkit - the one with 4 pads only. It's for xbox360, so I connect it via xbox 360 receiver + Santroller adapter (USB Inputs from xbox 360 wireless receiver -> Rock Band Drums). Works flawlessly but only 4 pads is boring.

sanjay900, 2025-02-11:
The thing that confuses me is lots of other people have had no issues with midi drums like this, so there's something specific going on here.

What have you got the minimum set to?

vladborovtsov, 2025-02-11:
> there's something specific going on here.
I agree.

The minimum value for all controls is 1024. I tried 0 and also something around 32k (which is the default half bar). This definitely isn’t a velocity or sensitivity issue. I also tried disabling the velocity sense on the drum kit.

Here’s some more information that might help:

When I hit a pad, it sends a **Note On** event. After about 11-40 milliseconds, it sends another **Note On** event for the same note, but with zero velocity.

I also tried the “fix note off” option, but I’m not sure what it does. It doesn’t seem to have any effect.

sanjay900, 2025-02-11:
i wonder if that note on with 0 velocity is the issue hmm

vladborovtsov, 2025-02-11:
Unfortunately I cannot test this. Midi cable offers to filter events by kind only, not by value. Though, there is a way to remap. I've tried to invert zero-velocity events, so those were coming with positive velocity value, so basically notes were doubled. No luck. Not better, not worse. 

Maybe there is a way to obtain some logs from the running Santroller firmware? I have seen there are some logging in the firmware src, but I dont see usb serial when connecting pico to Mac. Most likely for "release" builds those are turned off.

sanjay900, 2025-02-11:
Save your current configuration using the Save Settings button, and then send the following file:
```Windows: %appdata%/SantrollerConfigurator/Santroller/include/config_data.h
Linux: ~/.config/SantrollerConfigurator/Santroller/include/config_data.h
macOS: /Users/Shared/Library/Application Support/SantrollerConfigurator/Santroller/include/config_data.h```

sanjay900, 2025-02-11:
You need a second pico for debugging, youd wire it like below
```
pico A = the debugger
pico B = your midi pico

Pico A GND -> Pico B GND
Pico A GP4/UART1 TX -> Pico B GP1/UART0 RX
Pico A GP5/UART1 RX -> Pico B GP0/UART0 TX
```

vladborovtsov, 2025-02-12:
[config_data.h.zip](https://github.com/user-attachments/files/18764715/config_data.h.zip)

Sharing the `config_data.h`

Will try to capture UART output later today if it makes any sense.

sanjay900, 2025-02-12:
I'll need to send you a debug build, I'll do that tomorrow

sanjay900, 2025-02-14:
This one goes on your midi adapter
[firmware.uf2.zip](https://github.com/user-attachments/files/18805171/firmware.uf2.zip)

And this goes on the debug pico
https://github.com/raspberrypi/debugprobe/releases/download/debugprobe-v2.2.1/debugprobe_on_pico.uf2

vladborovtsov, 2025-02-15:
Tried it today. Here are the logs: 

[serial.log](https://github.com/user-attachments/files/18809990/serial.log)

Comparing the logs and analog output of the drum kit, I can see that the MIDI driver doesn't miss any beats. Everything is there. I've tried with different sensitivity - looks good. So N beats on drums - N*2 lines in the log. 

Your debug firmware works a bit differently than mine though. Sometimes it still may miss some beats, but it's much, much better. I’m even able to play double kicks and those are recognized by the game (Rockband 3 on PS3). 

Also, most of the time it still misses simultaneous notes, i.e red+green, red+blue, green+kick.  

On a separate note, 
I also have an Xbox 360 RB kit, the one without cymbals. It works absolutely OK through a Santroller adapter.

sanjay900, 2025-02-15:
The debug firmware is just the latest fw combined with that config,_data.h

sanjay900, 2025-02-15:
Usually, simultaneous notes is a debounce problem. The games have issues with simultaneous notes so the denounce actually seperates the notes out and only sends one note at a time.

vladborovtsov, 2025-02-15:
I thought debounce supposed to filter noisy inputs?

In the meantime, hitting buttons on the controller connected to another port to the hub causing no issues. 

And as I mentioned, xbox360 kits works fine through controller. So this makes me thing it's something about midi inout. 

Is there any way to make it better?

sanjay900, 2025-02-16:
You are on the latest santroller yes?

sanjay900, 2025-02-16:
Debounce for midi drums works a bit differently, because it has to due to some issues with how the games work
debounce gets used to both filter noisy inputs and isolate the inputs, as the RB games specifically have a bug where sending multiple inputs at once can cause issues, so the debounce gets reused to isolate the drum inputs.

You can test if this is the issue by emulating a GH kit, as we don't have that logic there.

sanjay900, 2025-02-16:
[firmware.uf2.zip](https://github.com/user-attachments/files/18813118/firmware.uf2.zip)
does this one do anything different

sanjay900, 2025-02-16:
[firmware.uf2.zip](https://github.com/user-attachments/files/18813132/firmware.uf2.zip)
or this one

sanjay900, 2025-02-16:
[firmware.uf2.zip](https://github.com/user-attachments/files/18813161/firmware.uf2.zip)
See if this makes a difference

vladborovtsov, 2025-02-16:
> [firmware.uf2.zip](https://github.com/user-attachments/files/18813161/firmware.uf2.zip) See if this makes a difference

Yes, it does. It works mostly similar to what I have got in the beginning. Simultaneous notes are working on this firmware, but even single notes are being missed. Though it obvious the drum kit itself is registering them. And also events are seen in the log. I'm sending a short video: https://drive.google.com/file/d/1Ok1-J65X--AbO8EL4v50UdT9iRE9YyC7/view?usp=share_link

So here its obvious- something is happening every N seconds so at least beat is missed.

sanjay900, 2025-02-16:
I sort of expect things to be dropped when we have logging set up though, since printing to the log takes a bit of time, might need to send a UF2 without the logging enabled to see if that makes a difference.

sanjay900, 2025-02-17:
[firmware.uf2.zip](https://github.com/user-attachments/files/18820673/firmware.uf2.zip)

sanjay900, 2025-02-17:
Also try v10.2.19 when it comes out, specifically seeing if with that one adjusting the debounce to a higher value helps at all.

vladborovtsov, 2025-02-17:
Just tried the firmware you shared recently. No difference at all compared to the one you shared before. 

Also tried 10.2.19 with different debounce values. Value between 40-65 works mostly the same and provides the most good experience.  If I disable debounce at all - thing is mostly not working missing 99% of beats. 

Though, there is a common observation with all debounce values as I mentioned before. 
Its feels like there some cycle or whatever (buffers processing?). The repetitive misses are being observed repeatedly and that interval is stable. 
Like, if I launch drum trainer with a simple rhythm, 3 yellow, red+yellow, 3 yellow, ..., I will always be missing one yellow, or, if I start plying in the middle, most likely red+yellow row will be always missed. 

Worth it to mention, the observed miss interval is the same regardless of BPM. So on 80 BPM I'll be missing 1 beat, on 140 BPM I'll be missing 1-2.

sanjay900, 2025-02-17:
There isn't any buffers on the firmware end, so I have no idea what's going on

sanjay900, 2025-02-17:
By the way, is there much of a difference if you set it as a gh kit instead of a rb one?

vladborovtsov, 2025-02-17:
> There isn't any buffers on the firmware end, so I have no idea what's going on

I was referring to this: https://github.com/Santroller/Santroller/blob/f42161e4dabd2b9f17090073fef68674b83b77d3/src/pico/midi_host.c#L41C1-L44C35 

Frankly speaking, I didn't put much efforts to get to the bottom of this. But the issue I'm experiencing could be related to a too small or too big buffer, depending on how they are consumed/implemented.

I'm judging from the realtime audio processing POV, but I think this is applicable to MIDI as well. 

Usually, shorter buffers => lower latency, but risk of glitches; for instance, since I'm observing a stable pattern of glitches with missing notes, my issue could be related to buffer underrun or overflow. So could be worth it to have an easy option to play with the constants I mentioned above. 

-- 

> By the way, is there much of a difference if you set it as a gh kit instead of a rb one?

Will try it and let you know.

sanjay900, 2025-02-17:
Those log events you see, those happen after that buffer
So if the logs are fine then that buffer probably isn't the issue

vladborovtsov, 2025-02-17:
Hmm, you are right!

Though, some other observations: 

I can see here https://github.com/Santroller/Santroller/blob/f42161e4dabd2b9f17090073fef68674b83b77d3/src/shared/hid/hid.cpp#L907 you are basically sampling velocities map? 

I think your initial assumption that drum kit triggers too quickly could be correct. If so - then this explains why debounce > 20 is so important to keep ball rolling. As I mentioned earlier, drum kit sends another note on event with zero velocity shorty, after 11-40ms. 

So it could be that note off (https://github.com/Santroller/Santroller/blob/f42161e4dabd2b9f17090073fef68674b83b77d3/src/shared/main/shared_main.cpp#L63) happens faster than velocities map is sampled during controller poll? 

there is usb handling, midi over usb handling, other stuff like emulation. I'm not sure at which rate it actually pools. I'm sure you may know better.

Here are some ideas.  

Drums nature is different than keys. While on keyboard you can press and hold a key, you cannot do so with the drum. You hit it and that's it.  You hit it again and that's another event for sure. 

Option 1: 
Instead of sampling maybe it worth it to use a switch? So note on received - you set velocity (or positive bool) in the map. And when you poll, you set it to 0/false (instead of memset). This will require mutex. Based on actual midi array poll rate, it may make things better or even worse (i.e. ghost/double/... beats). 

Option 2:
Use a FIFO approach (or almost FIFO). So instead of maintaining velocities array, maintain a queue. Note On - adds event to the FIFO queue, note off/zero velocity ignored. 
In the hid pooling function, you may have a "request to fulfill", imagine the worst case: 
```
struct {
   bool RED_PAD;
   bool YELLOW_PAD;
   bool BLUE_PAD;
   bool GREEN_PAD;
   bool YELLOW_CYMBAL;
   bool GREEN_CYMBAL;
   bool BLUE_CYMBAL;
   bool HIHAT;
   bool KICK;
}
```

So, for each struct member you scan the queue for the first matching event and pop it. 
Such approach may cause delay or even kinda "desync", but this will guarantee that all beats are delivered knowing MIDI driver doesn't miss anything. 

Those maybe a bit naive, but I'm just trying to solve my problem (and contribute!) as there is absolutely no way to get pro midi adapter for some reasonable price in my country.

sanjay900, 2025-02-17:
Tbh, we could just ignore the note off command and see if that fixes things, that should be enough if that is the actual problem.

vladborovtsov, 2025-02-17:
Well, you are maintaining the thing for a while. I'm sure all those problems aren't new. 

> Tbh, we could just ignore the note off command and see if that fixes things, that should be enough if that is the actual problem

I have seen note off fix in the configuration. Is this option exactly about ignoring note off? 

> The input processing loop clears the note as soon as it gets processed, so that should be good enough

mm, I believe you. That's almost what I suggested. But.. could it be the source of issue with missing one? If there is no queue, it then just clears note and that's it? 

--

On a separate note, if I take the config header built by configurator, could I compile the firmware using src from GitHub or there are some private parts? If it's buildable I'd be happy to play around implementing my ideas

sanjay900, 2025-02-17:
The thing is we poll every ms, and I doubt we get drum events that quickly right?

vladborovtsov, 2025-02-17:
of course right. I'm not that crazy to beat faster :D

sanjay900, 2025-02-17:
[firmware.uf2.zip](https://github.com/user-attachments/files/18833626/firmware.uf2.zip)
this one should be ignoring the note off event from the drums

vladborovtsov, 2025-02-17:
will test it tomorrow as well as GH kit emulation option. 
Note Off is considered the same as Note On with velocity = 0 ?

sanjay900, 2025-02-17:
correct, and its also ignoring any velocity < the velocity thats stored anyways too

sanjay900, 2025-02-18:
Alright we are getting somewhere, might just be missing from the config I was using

vladborovtsov, 2025-02-18:
I’m not sure if the configurator utility is reading the config file from the pico correctly. I checked, and the kick drum input was there and enabled. Should I wait for another binary or is the one you sent me reproducible using the most recent utility version?

sanjay900, 2025-02-18:
It's quite possible I just messed it up when I built it by hand

sanjay900, 2025-02-18:
What ill do is cut a new release, and then you can configure it all with  that and see what happens.

vladborovtsov, 2025-02-19:
I just tried setting up adapter using latest tool version you published today. 

There is still a problem with the kick. Now it is registered exactly one time (in the game). If I unplug and plug the device again to console - again, it will register the pedal trigger just once.

sanjay900, 2025-02-19:
Alright I can guess what's causing that, I'll fix it soon

vladborovtsov, 2025-02-19:
it seems it works! At least In trainer I'd doing 100% on 180 BPM. Will try playing a few songs and then will let you know.

closed 2025-02-19 by sanjay900

vladborovtsov, 2025-02-19:
@sanjay900 thank you so much for your help and support! The thing works really great!