Santroller/Santroller #277: [Issue]: Error on piano inputs

kimo1arquimedes1, 2025-02-19:
### Please describe your issue.

In this case I want to use a midi instrument (piano), the interface works perfectly with any music program. Unfortunately when adding it to the application there is no response, even though the usb midi interface cable flashes. What I can notice is that I had used a long time ago a rock band interface (keytar) and piano, and these were detected correctly, many versions ago. Currently I even connected these same devices and there was no response.

For example: I connect the keytar (rock band) and the keys, it detects them perfectly with usb host, and then I connect the midi interface and it is even detected as -midi-, when adding midi inputs the option to click on the 1st key appears, but none of them work. As I said, it worked before and I have already tried with two pianos and two interfaces and there is no response.

### Version

v10.2.20 

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-02-19:
i assume things work on v10.2.18 right

sanjay900, 2025-02-19:
Should be fixed with v10.2.21 as well when that comes out

kimo1arquimedes1, 2025-02-19:
The problem has been solved, thank you very much.

closed 2025-02-19 by sanjay900