Santroller/Santroller #279: [Issue]: Feather RP2040 USB Host on Xbox One resets once controller input is sent, losing authentication

MouseRB, 2025-02-26:
### Please describe your issue.

Hi, I'm trying to use a Adafruit Feather RP2040 USB Host as an adapter to use instruments on different consoles and I've noticed that on Xbox One, I can authenticate the controller (looking at Rock Band 4's main menu, the Guitar icon appears and the option to push start appears), but as soon as I press anything on the controller the icon disappears and disconnects. I've tried with an Xbox One Riffmaster, and authenticating with an Xbox One controller, unplugging it, and plugging in an Xbox 360 controller with the same issue happening both ways.

I've also tried the adapter on an Xbox 360 and it works as expected there, even using a wired controller to authenticate and then unplugging and using the Riffmaster.

### Version

10.2.23

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-02-27:
very weird, haven't had anyone report this issue yet

MouseRB, 2025-03-01:
Is there anything you can suggest trying on my end? I tried a USB hub to connect both together but this caused the controller to never connect at all.

sanjay900, 2025-03-01:
Try the latest release first 
If that doesn't work try older ones and see if any of them work

MouseRB, 2025-03-01:
Updated to the latest release and rebuilt the adapter from scratch (holding down the boot button so it "clean" resets), and that seems to have fixed it, thanks!

closed 2025-03-01 by MouseRB