Santroller/Santroller #281: Error building

Myxelium, 2025-03-03:
Santroller Version: 10.2.29
OS Version: 10.0.22631.0
OS Platform: Win32NT
OS Service Pack: 
OS VersionString: Microsoft Windows NT 10.0.22631.0

Device Type: GuitarHeroGuitar
Led Type: None

Microcontroller Type: Raspberry PI Pico
Microcontroller Frequency: 125000mhz

Processing picow (board: rpipicow; platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipicow.html
PLATFORM: Raspberry Pi RP2040 (1.15.0) > Pico W
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40404.0 (4.4.4) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 71 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.3
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\picow\FrameworkArduinoVariant\digital.cpp.o
Compiling .pio\build\picow\FrameworkArduinoVariant\init.cpp.o
Compiling .pio\build\picow\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\picow\src\pico\bt.cpp.o
Compiling .pio\build\picow\src\pico\bt_rx.cpp.o
Compiling .pio\build\picow\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\picow\src\pico\controller_simulator.c.o
Compiling .pio\build\picow\src\pico\hidparser.c.o
Compiling .pio\build\picow\src\pico\io.cpp.o
Compiling .pio\build\picow\src\pico\main.cpp.o
Compiling .pio\build\picow\src\pico\midi_host.c.o
Compiling .pio\build\picow\src\pico\pins.cpp.o
Compiling .pio\build\picow\src\pico\rosc.c.o
Compiling .pio\build\picow\src\pico\sleep.c.o
Compiling .pio\build\picow\src\pico\xinput_device.cpp.o
Compiling .pio\build\picow\src\pico\xinput_host.cpp.o
Compiling .pio\build\picow\src\shared\config\config.c.o
Compiling .pio\build\picow\src\shared\controllers\controllers.c.o
Compiling .pio\build\picow\src\shared\hid\hid.cpp.o
Compiling .pio\build\picow\src\shared\main\accel.cpp.o
Compiling .pio\build\picow\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\picow\src\shared\main\max1704x.cpp.o
Compiling .pio\build\picow\src\shared\main\mpr121.cpp.o
[... 102 lines cut ...]
Compiling .pio\build\picow\FrameworkArduino\wiring_shift.cpp.o
Archiving .pio\build\picow\libFrameworkArduino.a
Linking .pio\build\picow\firmware.elf
Generating UF2 image
picotool uf2 convert -t elf ".pio\build\picow\firmware.elf" ".pio\build\picow\firmware.uf2"
Retrieving maximum program size .pio\build\picow\firmware.elf
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
Checking size .pio\build\picow\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.5% (used 119176 bytes from 262144 bytes)
Flash: [==        ]  22.3% (used 466480 bytes from 2093056 bytes)
Building .pio\build\picow\firmware.bin
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = mbed
Trying to reset Pico into bootloader mode...
Looking for upload disk...
Using manually specified: N:\
Uploading .pio\build\picow\firmware.bin
*** [upload] Invalid argument
========================= [FAILED] Took 28.43 seconds =========================

sanjay900, 2025-03-03:
Try the portable version

Myxelium, 2025-03-03:
using the portable version

sanjay900, 2025-03-03:
Okay then, try the not portable version

Myxelium, 2025-03-03:
pico keeps disconnecting and reconnecting nonstop. its a new one out of the box

sanjay900, 2025-03-03:
oh
that sounds like an issue with the pico then
have you tried putting it on bootsel mode yourself?

sanjay900, 2025-03-03:
https://santroller.tangentmc.net/tool/rescuing.html

sanjay900, 2025-03-03:
if its disconnecting and reconnecting non stop when its in that mode, then i would have to assume something is broken with the pico unfortunately, no idea what would cause that

Myxelium, 2025-03-03:
got it to write multiple times at first then usb issues came meanwhile the bluetooth worked flawlessly at the same time it reconnected over usb, ive ordered 2 new ones

Myxelium, 2025-03-03:
bootsel doesnt do anyting except stopping it from connecting at all

Myxelium, 2025-03-03:
To be honest after 2 broken boards and 10 passes through all connections on my board, I genially believe there is something suspect with Santroller. Checked all connections if any were wired to wrong pin or if any was shorted, but nope. After being programmed once, it completely goes into crazy mode and refuses to stay connected.

This is my wiring:

[image]

PS2 - World tour guitar (no slider connection or joystick, no tilt)

sanjay900, 2025-03-03:
does it happen with nothing wired up

Myxelium, 2025-03-03:
Same as listed, joystick haven't been wired for. It happens even when nothing is connected to any port except micro usb. It opens the folder with the files in and a moment later disconnect happens and it closes it. However this might be the original triggers causing shorts?

For whammy, i use 3.3v gp26 and gnd. it gets detected and show movement in santroller the seconds I get to see it before it disconnects.

the picos are bought from amazon: https://www.amazon.se/dp/B0B5H17CMK?ref=ppx_yo2ov_dt_b_fed_asin_title

Both are now completely undetectable on my windows machine. I plugged them out of the wiring (sockets) and they disconnected and reconnected a few times and now they dont show any sign of life at all.

conf: https://filebin.net/ivpwcgxj2x3m540b/controller.picoconfig

sanjay900, 2025-03-03:
which 3.3v rail are you using

sanjay900, 2025-03-03:
If it was a santroller issue, then holding bootsel and plugging the pico in would work just fine for making it show up.
If that isnt working, then i would assume something is borked with the specific picos you have.

sanjay900, 2025-03-03:
okay so linux requires the driver to pick the device up after programming, but won't require it for the bootsel part
windows only needs drivers for the arduinos

Myxelium, 2025-03-03:
which 3.3v rail are you using: 36

I think I might give this up, I have one new unused pico left but it will probably also get bricked if i try to get this working

sanjay900, 2025-03-03:
Its very odd, the picos shouldn't actually be brickable since the bootloader is in the silicon itself, so i don't understand whats going on at all

sanjay900, 2025-03-03:
Can you hit save settings and then send a copy of one of these?
Windows: `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`
Linux: `~/.config/SantrollerConfigurator/Santroller/include/config_data.h`

Myxelium, 2025-03-03:
No such file, config.h exists

sanjay900, 2025-03-03:
itll only exist right after hitting save settings

Myxelium, 2025-03-03:
Well, I usually get to do that twice or so before the Pico goes wild and starts its reconnecting disconnecting journey. 
That's why I had my settings saved as a preset because I tried being quick about programming it and going full bt once its programmed.

I'm doing a complete rewiring tomorrow and use a regular pico as a guinea pig, hopefully something changes.

Myxelium, 2025-03-08:
Update, I still haven't tried it in my guitar yet since ive been sick a week. However i managed to do the Startup guide from raspberry's website on one of the "bricked" pi's. I opened my old linux machine and installed micropython, after that windows now recognises without the disconnect / reconnecting happening. 

The pico seems to be working normally, got the [blink webserver ](https://projects.raspberrypi.org/en/projects/get-started-pico-w) running without any issues, from their tutorial. So, I managed to unfuck one of the 2 bricked ones.

May this has something to do with programming through test pins, as i had usb c upgrade done? It has also been connected to the guitar switches, buttons and whammy arm when programming has been made in the past, ive known that causes issues in some arduinos, might also be the case here?

[image]

Myxelium, 2025-03-08:
I can happily call this issue solved. It worked flawlessly now after downloading latest version with the one PICO that started working. 

I'm following your guide, however I'm a bit confused what more I should do with the MPR121 chip. The guitar hero controller gives me 1 wire to work with for the slider pad.

So far I've connected it to the pin 1 of MPR121 I'm a bit confused with the channel text in the guide. 

I've put a line under the text that confuses me a bit, 

[image]

closed 2025-03-08 by Myxelium

sanjay900, 2025-03-08:
Follow the wiring guide first before the programming guide