Santroller/Santroller #287: [Issue]: MicrokorgXL cannot be used as Midi device

direbedlam, 2025-03-26:
### Please describe your issue.

Hi there, love your app, having a lot of fun with the Wii Guitar I converted to use on XBox 360.

My issue pertains to the use of the Korg MicroKorg XL keyboard for use with the Midi adapter.  I have confirmed that my pi pico with the ProKeys config works on my Roland upright piano, but it doesn't work with the MicroKorgXL no matter what configuration I change.  My guess is that the issue arises from the fact it appears to have two midi input options when I connect it directly to my computer.  Only one of those inputs works for reading the key strokes.  Have you experienced this issue before and do you have any advice on how I might get around it.  I'm considering getting a MIDI to USB adapter to see if that will work instead of the native USB connection.

### Version

Latest

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-03-28:
I suspect a MIDI to usb adapter would work, I haven't actually encountered this problem before since most MIDI devices don't work like that.

Would you be able to get a descriptor dump for it using https://www.thesycon.de/eng/usb_descriptordumper.shtml?

sanjay900, 2026-10-13:
The new firmware supports more devices, if this still doesnt work feel free to reopen

closed 2026-10-13 by sanjay900