Santroller/Santroller #288: [Issue]: Santroller stuck on

whtrmshrms, 2025-03-26:
### Please describe your issue.

Hey! Thank you so much for your hard work. I have a problem, and hopefully you maybe able to assist. Initial programming of the controller works fine, and I can play it. I just can't make changes to the controller anymore, nor can I reset back to Arduino using the configuration tool. I switched from a ball tilt to MPU 6050. When I make an adjustment to the controller and click save, it'll remain stuck at 0% "Erasing Device- Building". When I try to reset back to Arduino, it'll get to 10% but stay stuck on "Entering Programing mode". Thank you, appreciate any input! 

### Version

10.3.15

### What operating system are you using?

Windows 10

### os-other

n/a

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2025-03-26:
Downgrade to 10.1.188, there's issues with the pro micro right now

closed 2025-03-26 by sanjay900

dayeggpi, 2025-06-23:
@sanjay900 having same issue right now... I tried with latest release, as well as [v10.3.52](https://github.com/Santroller/Santroller/releases/tag/v10.3.52) (just randomly), do i need to go back to 10.1.188 ? If supposed to be fixed I can open a new ticket and provide any data you need from me, just let me know where to find some logs.