Santroller/Santroller #293: [Issue]:

Vaiski06, 2025-05-01:
### Please describe your issue.

rocessing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Downloading 0% 10% 20% 30% 40%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Cannot read termcap database;
using dumb terminal settings.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.15.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40404.0 (4.4.4) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 71 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.3
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio/build/pico/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o
Compiling .pio/build/pico/src/pico/bt.cpp.o
Compiling .pio/build/pico/src/pico/bt_rx.cpp.o
Compiling .pio/build/pico/src/pico/bt_rx_cnt.cpp.o
Compiling .pio/build/pico/src/pico/controller_simulator.c.o
In file included from include/config.h:10,
                 from src/pico/bt_rx_cnt.cpp:2:
include/config_data.h:18:19: error: extended character − is not valid in an identifier
   18 | #define SLEEP_PIN −1
      |                   ^
In file included from include/config.h:10,
                 from src/pico/bt_rx.cpp:1:
include/config_data.h:18:19: error: extended character − is not valid in an identifier
   18 | #define SLEEP_PIN −1
      |                   ^
In file included from include/config.h:10,
                 from src/pico/bt.cpp:1:
include/config_data.h:18:19: error: extended character − is not valid in an identifier
   18 | #define SLEEP_PIN −1
      |                   ^
*** [.pio/build/pico/src/pico/bt.cpp.o] Error 1
*** [.pio/build/pico/src/pico/bt_rx.cpp.o] Error 1
*** [.pio/build/pico/src/pico/bt_rx_cnt.cpp.o] Error 1

### Version

v10.3.47

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-05-01:
what language is your pc set to?

sanjay900, 2025-05-01:
Does finnish use a different style of dash for negative numbers?

sanjay900, 2025-05-03:
Fixed with v10.3.50

closed 2025-05-03 by sanjay900