Santroller/Santroller #296: [Issue]: stuck at 18% [ Status: Writing - Building ] then issue pops up

putimadri, 2025-05-07:
### Please describe your issue.

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Downloading...
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
HTTPClientError: 

I tried using different users within the pc and dint work, changed the file location of santroller to the root of my C drive and even tried on a separate pc and none of these worked, very confused as i used santroller to program a different controller no longer than 2 weeks ago and it worked perfectly fine then

### Version

10.4.3

### What operating system are you using?

Windows 10

### os-other

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-05-07:
This is odd it's having issues downloading some libraries it seems

sanjay900, 2025-05-07:
Is there more to that error

closed 2025-05-07 by sanjay900