Santroller/Santroller #298: [Issue]:

toxicRavioli, 2025-05-08:
### Please describe your issue.

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Downloading...
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Downloading 0% 10% 20% 30% 40%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.15.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40404.0 (4.4.4) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 71 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.3
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\pico\src\pico\controller_simulator.c.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\rosc.c.o
Compiling .pio\build\pico\src\pico\sleep.c.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
In file included from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/assert.h:14,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/types.h:12,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico.h:26,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:10,
                 from src\pico\pins.cpp:1:
C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/stdlib.h:36,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico\cores\rp2040/Arduino.h:24,
                 from src\pico\main.cpp:1:
C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\pico\pins.cpp.o] Error 1
In file included from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/assert.h:14,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/types.h:12,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico.h:26,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_spi/include/hardware/spi.h:10,
                 from src\pico/io_define.h:3,
                 from include/io.h:4,
                 from src\pico\io.cpp:1:
C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\pico\main.cpp.o] Error 1
*** [.pio\build\pico\src\pico\io.cpp.o] Error 1
In file included from C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/stdlib.h:36,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico\cores\rp2040/Arduino.h:24,
                 from .pio\libdeps\pico\USB-Host-Shield-20/version_helper.h:31,
                 from .pio\libdeps\pico\USB-Host-Shield-20/settings.h:100,
                 from .pio\libdeps\pico\USB-Host-Shield-20/Usb.h:34,
                 from include/descriptors.h:4,
                 from src\pico\xinput_device.cpp:11:
C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/stdlib.h:36,
                 from C:\Users\Patrick\OneDrive\Desktop\New folder\SantrollerConfigurator-win-Portable (1)\current\Binaries\platformio\packages\framework-arduinopico\cores\rp2040/Arduino.h:24,
                 from .pio\libdeps\pico\USB-Host-Shield-20/version_helper.h:31,
                 from .pio\libdeps\pico\USB-Host-Shield-20/settings.h:100,
                 from .pio\libdeps\pico\USB-Host-Shield-20/Usb.h:34,
                 from include/descriptors.h:4,
                 from src\pico\xinput_host.cpp:32:
C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)/current/Binaries/platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\pico\xinput_device.cpp.o] Error 1
*** [.pio\build\pico\src\pico\xinput_host.cpp.o] Error 1
========================= [FAILED] Took 12.67 seconds =========================

### Version

10.4.4

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-05-08:
`C:/Users/Patrick/OneDrive/Desktop/New folder/SantrollerConfigurator-win-Portable (1)`
remove the spaces from `new folder` and `SantrollerConfigurator-win-Portable (1)`

toxicRavioli, 2025-05-08:
got it thank you brother!