Santroller/Santroller #304: [Issue]: Erro ao programar o microprocessador

AJHERO, 2025-05-25:
### Please describe your issue.

[image]

### Version

10.4.24

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-05-25:
can you copy paste the entire error?

AJHERO, 2025-05-25:
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Downloading...
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
HTTPClientError:

sanjay900, 2025-05-25:
theres a problem with your internet connection, its blocking access for downloading libraries that are being updated