Santroller/Santroller #306: [Request]: PS4 key-based authentication

SolidElectronics, 2025-05-27:
### Please describe your request.

I was reading the documentation for [GP2040-CE](https://gp2040-ce.info/web-configurator/menu-pages/settings#additional-ps4-settings), and they've found a way to implement PS4 authentication without needing a passthrough to a physical controller.  It's one of those situations where you're supposed to dump your own Dualshock 4 firmware and extract the keys yourself, and they won't supply them for you.  Is this something that would be possible to implement in Santroller?

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2025-05-27:
It's one of those things where I could implement it but then it only really works for PS4 and it's not a simple process to dump said keys either

sanjay900, 2025-05-27:
For RB its much easier to just use PS3 instruments and not deal with auth

closed 2025-06-24 by SolidElectronics