Santroller/Santroller #320: [Issue]: Stuck at "Status: Writing - Looking for device"

N1GHTOUWL, 2025-07-14:
### Please describe your issue.

I'm trying to set up my microcontroller so I can start testing where each wire should be connected to the pico. Whenever I first launch santroller, I go through the motion of changing the emulation from gamepad to gh controller and configuring. But after that I just stays stuck at 0%. I don't know that I am doing wrong. 

### Version

v10.4.53

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-07-14:
odd, are you able to test it on a second computer?

N1GHTOUWL, 2025-07-14:
Unfortunately not. Would maybe deleting the application and redownloading it make it work? Also, SantrollerConfigurator-win-Setup.exe is the correct download right?

sanjay900, 2025-07-14:
yeah, though you can try the portable version as well and see if anything changes

N1GHTOUWL, 2025-07-14:
Also I do have another question pertaining the pico. I'm playing on wiring everything to the pico and running a usb cable to the pc. But whenever I try looking through your guide, you state how "Needs to be a wired controller, wireless only controllers won't work, nor will the play and charge kit." I'm converting a wireless ps2 gh controller and wanted to know if that would work and which steps should I follow in your guide? or any other guides or instructions you could show me that would help? It's my first time messing with this type of thing.

sanjay900, 2025-07-14:
https://santroller.com/wiring_guides/guitar.html
you want to follow this guide for wiring, the notes about wireless controllers are only for USB host and that is only used for auth

N1GHTOUWL, 2025-07-15:
Could it be the way I configured the pico? Is there a certain package that I have to download specifically for the santroller to read the pico?

sanjay900, 2025-07-15:
no, it should just work, does your username on your pc have a space in it? that can throw it off sometimes

N1GHTOUWL, 2025-07-15:
Weird. This pico just doesn't want to be read. But it does read it before pressing configure. It shows that its reading a pico but after that it's like it disappears.

sanjay900, 2025-07-15:
what do you see in device manager when its disappeared?

N1GHTOUWL, 2025-07-15:
Ok so I reset it again while santroller was still trying to find it and it finally loaded to 100 and allowed me to configure. The pico currently un-updated and without a package installed.

N1GHTOUWL, 2025-07-15:
My guess is when I updated the firmware or installed a package, santroller couldn't read or find the pico. But after resetting it, it allowed santroller to find and install/ configure what was needed