Santroller/Santroller #331: [Issue]: "Failed to read data from device: Operation timed out" when programming buttons.

sync-on-luma, 2025-09-02:
### Please describe your issue.

I'm trying to set up a microcontroller to emulate an Rock Band guitar for the Xbox 360 .  I have a Arduino Mega 2560 clone from SainSmart. I'm aware there are potential compatibility issues with Arduino clones, however this one does not have the ch340g chip that was mentioned as causing problems, and appears to be identical to an official board.

The initial programming of the controller went fine, so I mounted it in my guitar chassis and began assembling everything.  After getting the frets hooked up I decided to test the IO to make sure everything worked, and this is where I ran into trouble. 

The controller is detected by santroller, and I can get to the input editor just fine, but when I try to detect a pin, nothing happens. If I launch the application from the terminal and try the same thing I get the output "Failed to read data from device: Operation timed out". This happens whenever any IO pin is bridged to ground, analog or digital. Running the application as root also has no effect.

I also tried manually setting each pin and saving the configuration. That seemed to work, but the buttons still do nothing, even though the controller is detected as an Xinput device by everything I plug it in to.

Does anyone know what might be going on here, or do I just have an incompatible clone?

### Version

10.4.75

### What operating system are you using?

Linux

### What microcontroller are you using?

arduino mega 2560

### How is your controller programmed?

Direct

closed 2025-09-02 by sanjay900

sanjay900, 2025-09-02:
https://github.com/Santroller/Santroller/releases/tag/v10.1.188

sanjay900, 2025-09-02:
I am going through and making it a lot more clear that these devices aren't really supported by modern versions of the tooling, and that future versions of the tooling won't support it at all.

sync-on-luma, 2025-09-02:
> https://github.com/Santroller/Santroller/releases/tag/v10.1.188

Alright, I've been messing around with this version for a couple hours now and I can't get anywhere with it. On Linux it won't even check for devices due to expecting an older version of libusb, and on Windows the writing process gets stuck at 90% every time. This happens on both a VM and real hardware.
The message says to try clicking "Refresh Devices" if it gets stuck, but there's no option to do that?

sync-on-luma, 2025-09-02:
> You probably need to follow the [rescue instructions](https://santroller.com/tool/rescuing.html) to nuke any other version of santroller first

I've already tried entering DFU mode to program it and had the same result. I'll try one more time though just to make sure.

sanjay900, 2025-09-02:
whats it showing up as in device manager when its stuck at 90%?

sync-on-luma, 2025-09-02:
Yeah it's still doing the same thing.
Device manager shows USB serial device (COM4). I'm pretty sure that's it.

sanjay900, 2025-09-02:
if you restart santroller at that point, what does it think the mega is?

sync-on-luma, 2025-09-02:
Arduino Mega 2560 (COM4)

sanjay900, 2025-09-02:
whats the vid and pid of it?

sync-on-luma, 2025-09-02:
VID 2342
PID 0042

sync-on-luma, 2025-09-02:
I select the Arduino type and hit Configure. Then it says "Building" for about a second and jumps straight to 90%

sanjay900, 2025-09-02:
okay yeah thats not right at all hmm
and it does that even when you start from DFU mode too?

sync-on-luma, 2025-09-02:
Yeah same thing in DFU mode. It maybe hangs for an extra second first.

sync-on-luma, 2025-09-02:
God dammit.
The whole reason I'm using this thing is because I already had it and had no other use for it. I do have I Pico, but I need for Xbox modding.
I really wanted to get this done today. Not to mention all the extra soldering I'll have to do now.

sync-on-luma, 2025-09-02:
> There is one thing you can try though Using 10.4.75, try writing a config for a pin really quickly. The arduinos are reponsive for a moment after you start configuring, but after a little bit they just stop accepting commands

What do you mean by really quickly? Like plug it in and  write the config immediately afterwards

sanjay900, 2025-09-02:
> What do you mean by really quickly? Like plug it in and write the config immediately afterwards

Yeah. In theory, things like pin detection should work for a moment. Actually let me look for a different version to try as well, one sec

sync-on-luma, 2025-09-02:
Just tried it. Unplugging the board kick me back to the menu, and I can't get back into the configurator and select an input fast enough for whatever tiny timing window it may have.

sanjay900, 2025-09-02:
[v10.3.8](https://github.com/Santroller/SantrollerConfigurator/releases/tag/v10.3.8)
try this

sync-on-luma, 2025-09-02:
I don't see 10.3.8. It goes straight from 10.3.52 to 10.4.0

sanjay900, 2025-09-02:
https://github.com/Santroller/Santroller/releases/tag/v10.3.8

sync-on-luma, 2025-09-02:
> https://github.com/Santroller/Santroller/releases/tag/v10.3.8

That worked a lot better, but it still seems to be having issues. After re-flashing the board, there's no more time out error, but trying to detect a pin still doesn't do work. Now it just makes the orange light on the board blink until the program is closed.

sanjay900, 2025-09-02:
does configuring pins manually do anything different with that one?

sync-on-luma, 2025-09-02:
Holy crap that worked!
It was incredibly tedious, but it worked.

sync-on-luma, 2025-09-02:
It's not working with the Xbox, but that might be an issue with the USB patch. I thought I set it up correctly, but now I'll have to double check.

At least I know the guitar is working in some capacity. I'm gonna finish assembling it and I'll let you know if I run across any more issues.

sync-on-luma, 2025-09-02:
Attempting to do the final controller program after setting all the pins but it keeps crashing on save.
I have to fully reprogram the controller every time it fails like this.

Here's the output
`Processing arduino_mega_2560 (board: megaatmega2560; platform: atmelavr; framework: arduino)
--------------------------------------------------------------------------------
Cannot read termcap database;
using dumb terminal settings.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PsxNewLib @ 0.4.0
|-- USB-Host-Shield-20 @ 1.6.0
|-- SPI @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio/build/arduino_mega_2560/src/avr/shared/io.cpp.o
Compiling .pio/build/arduino_mega_2560/src/avr/shared/pins.cpp.o
Compiling .pio/build/arduino_mega_2560/src/avr/shared/ps2_emu.cpp.o
Compiling .pio/build/arduino_mega_2560/src/avr/uno/main/main.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/config/config.c.o
Compiling .pio/build/arduino_mega_2560/src/shared/controllers/controllers.c.o
Compiling .pio/build/arduino_mega_2560/src/shared/hid/hid.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/main/accel.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/main/max1704x.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/main/mpr121.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/main/ps2.cpp.o
In file included from include/config.h:10:0,
                 from src/avr/shared/pins.cpp:6:
include/config_data.h:38:26: error: narrowing conversion of '-1' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
 #define ADC_PINS {0,-1,-2}
                          ^
src/avr/shared/pins.cpp:12:39: note: in expansion of macro 'ADC_PINS'
 const uint8_t analogPins[ADC_COUNT] = ADC_PINS;
                                       ^~~~~~~~
include/config_data.h:38:26: error: narrowing conversion of '-2' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
 #define ADC_PINS {0,-1,-2}
                          ^
src/avr/shared/pins.cpp:12:39: note: in expansion of macro 'ADC_PINS'
 const uint8_t analogPins[ADC_COUNT] = ADC_PINS;
                                       ^~~~~~~~
Compiling .pio/build/arduino_mega_2560/src/shared/main/shared_main.cpp.o
Compiling .pio/build/arduino_mega_2560/src/shared/main/wii.cpp.o
*** [.pio/build/arduino_mega_2560/src/avr/shared/pins.cpp.o] Error 1
src/shared/main/shared_main.cpp: In function 'int16_t handle_calibration_xbox_whammy(int16_t, uint32_t, uint32_t, int16_t, uint16_t)':
src/shared/main/shared_main.cpp:360:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (val > min) {
             ~~~~^~~~~
src/shared/main/shared_main.cpp: In function 'uint16_t handle_calibration_xbox_one_trigger(uint16_t, uint16_t, uint32_t, int16_t, uint16_t)':
src/shared/main/shared_main.cpp:386:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (val > min) {
             ~~~~^~~~~
src/shared/main/shared_main.cpp: In function 'uint16_t handle_calibration_drum(uint16_t, uint16_t, uint32_t, int16_t, uint16_t)':
src/shared/main/shared_main.cpp:412:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (val > min) {
             ~~~~^~~~~
src/shared/main/shared_main.cpp: In function 'uint8_t tick_inputs(void*, USB_LastReport_Data_t*, uint8_t)':
src/shared/main/shared_main.cpp:2307:20: warning: unused variable 'current_drum_report' [-Wunused-variable]
     USB_RB_Drums_t current_drum_report = {buttons : 0};
                    ^~~~~~~~~~~~~~~~~~~
src/shared/main/shared_main.cpp:2308:10: warning: unused variable 'drumHit' [-Wunused-variable]
     bool drumHit = false;
          ^~~~~~~
src/shared/main/shared_main.cpp:2310:10: warning: unused variable 'drumSeen' [-Wunused-variable]
     bool drumSeen[DIGITAL_COUNT] = {false};
          ^~~~~~~~
src/shared/main/shared_main.cpp:2457:10: warning: unused variable 'updateSequence' [-Wunused-variable]
     bool updateSequence = false;
          ^~~~~~~~~~~~~~
src/shared/main/shared_main.cpp: At global scope:
src/shared/main/shared_main.cpp:53:3: warning: 'rtt_t' defined but not used [-Wunused-variable]
 } rtt_t;
   ^~~~~
src/shared/main/shared_main.cpp:42:3: warning: 'ltt_t' defined but not used [-Wunused-variable]
 } ltt_t;
   ^~~~~
========================== [FAILED] Took 0.92 seconds ==========================
`

sanjay900, 2025-09-02:
Which analog pins are you using?
It doesn't seem to like one of them

sync-on-luma, 2025-09-02:
> The only working analog pin is pin a0 right now

That explains a lot. I have three analog pins connected. One for the whammy bar, one for the pickup selector, and one because it was the easiest point for that button to reach.

sync-on-luma, 2025-09-02:
Question: what's the 360 auth keys thing in the programmer? I couldn't find anything about it in the documentation.
 I tried importing the keys from my xbox with it, but it didn't seem to do anything.

sync-on-luma, 2025-09-02:
Also the controller still isn't connecting to the Xbox. I've double checked an the plugin is set up correctly in Dashlaunch

sanjay900, 2025-09-02:
> Question: what's the 360 auth keys thing in the programmer? I couldn't find anything about it in the documentation. I tried importing the keys from my xbox with it, but it didn't seem to do anything.

not compatible with arduinos, thats a pico only thing.

sanjay900, 2025-09-03:
ah, yeah good that you have something else to check with

sanjay900, 2025-09-03:
An xb1 controller wouldn't work, it needs to be something xinput and xb1 controllers aren't.