Santroller/Santroller #333: [Issue]: Turntable platter setup broken?

dominuick, 2025-09-07:
### Please describe your issue.

When setting up my turntable I get an error on the turntable setup

I choose pins 18 and 19 as the guide suggests, but I get a missing pin config error, and never get a left or right platter connected

If I try to proceed with flashing with the error, I get a broken firmware that doesn't work, and need to reset the pico

I've tried older versions of santroller

### Version

10.5.1

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-09-07:
Out of curiosity, does it let you program it fine if the turntables aren't connected?
In the past when this sort of thing has happened it's usually a wiring problem

dominuick, 2025-09-07:
> What happens if you program it for pins that aren't connected to anything That would confirm if it's a wiring problem

Messed with it for a while, santroller still gives me errors, but everything including the platter works when plugged into my 360

Probably just my old laptop and wierd windows install

sanjay900, 2025-09-07:
Yeah the errors are a bug, I should fix them but im too busy rewriting everything which will solve all these issues, good to hear you got it working