Santroller/Santroller #335: [Issue]: Rockbadn 2 Wii Drum Cymbals detected as pad on PS4/PS5 Mode

profilku, 2025-09-14:
### Please describe your issue.

Hello, I have an issue using Rockband 2 Wii Drums with cymbals connected through USB Host
Peripheral : Wii Rockband 2 drum with cymbals
Console : PS4
Game : Rockband 4

When Emulation mode for PS4/PS5 Mode disabled (PS3 mode) :
Cymbals detected as cymbals , all works well

Emulation mode for PS4/PS5 Mode enabled :  
Cymbal detected same as tom in the game

### Version

10.5.5

### What operating system are you using?

Mac OS

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-09-15:
I'll have a look when I get some time

sanjay900, 2025-11-30:
I suspect this is probably solved in newer releases